polvinc/HEMY #1: Has HEMY ever been actually made?

martin0359, 2024-10-03:
Hi Polvinc, HEMY looks like the better design, but DHEM offers a more cost-effective way of thinking about it. I would like to know if the dimensions of the HEMY parts are accurate and I would like the CNC machining to be done in one time. If they are not verified, should I make the DHEM first?I already have a tiny experience of making a 3d printed Keen eq mount once.

polvinc, 2024-10-05:
That's a good question! 
In practice, HEMY outperforms DHEM. I was able to test the latter this summer, and I made two of them. I'm achieving tracking performance of less than 1 arcsec with a load of around 13kg, with no counterweight. Here's an example image taken with HEMY (I don't have any excellent astrophoto equipment, just a Nikon D750 that I defiltered myself, and a Sigma 150-600mm; for this image I used 50x5min exposure, ISO1600, 400mm f7.1, with DOF). 
[image]<br /><a target='_blank' href='https://imgbb.com/'>IC1396</a><br />

Following these tests, I made a few minor modifications to HEMY mount, which are worth waiting for the plans to be updated before starting to build it. I'm also in the process of developing HEMY's electronics, which will be integrated into it, and above all will be able to manage the brake (which is not currently the case with separated onstep). The development will be ready in two or three months (joint work with an acquaintance, https://github.com/zUrp-Astronomics/unicorn).

In the meantime, if this is your first DIY project with this level of investment, I'd recommend starting with DHEM. You can use the same gearboxes as the HEMY (on aliexpress). You can then use them for the HEMY. 

It's taking me a while to reply, but don't hesitate to ask questions, even simplistic ones!

martin0359, 2024-10-06:
Thank you, Polvinc, I couldn't resist CNC all the parts of the HEMY main body when i waiting for your reply, and I expect to receive it after 7 days.The AZ BASE part I am going to buy the finished product, I'm in China and expect the overall cost to be much lower than you anticipate.I'd also like to test some harmonic driver that made in China.I have studied the drawings carefully and like the HEMY design even more. 

More questions will be communicated to you during the build, thanks again for your great design. By the way do the new improvements require all the parts to be reCNC?

martin0359, 2024-10-06:
No. 11, “diameter 8 shaft”, what is this part made of and what is its length?

martin0359, 2024-10-06:
Also the bom list in readme.md seems to have a different motor model than in the pdf and is missing the belt.
Which is the necessary setup?

polvinc, 2024-10-06:
Glad to hear you're getting into manufacturing! I'd love to hear your feedback on assembly and performance testing!

No, the improvements do not require any machining modifications to the frame. The main changes concern the ALT/AZ base.
On the other hand, as this was the first build, I took quite a few liberties in adjusting the coupling between the harmonic gearbox and the motor. In particular, the spacing between the wave generator and the bearing. It's currently missing a stop at the right dimension.
I used M8 wide washers for this, as well as shim-seal washers:
https://fr.aliexpress.com/item/1005001481545326.html
https://fr.aliexpress.com/item/1005002374937009.html

To assemble, attach the shaft (phi 8mm, length 40mm, https://fr.aliexpress.com/item/1005006559769183.html) to the wave generator. This assembly is inserted into the flex spline of the harmonic gearbox, with the shaft pressure screws on the gearbox side.
Then fit the backlash adjustment washers (one M8x24x2 wide, and four M8x14x0.1mm). The wide washer will be used to support the bearing. Clearly, this part will eventually be replaced by a machined part.
Then, this assembly (complete gearbox + shaft + washers) is inserted into the bearing with part number 9 on the drawing, and the assembly is locked in translation with the pulley.
I use pulleys:
- For RA: 80-tooth pulley, 180-tooth GT2 belt, 16-tooth motor side
- For DEC: 60-tooth pulley, 150-tooth GT2 belt, 16-tooth motor side

As for the engines, I haven't bothered for the moment, I've taken two that I had in stock: https://www.omc-stepperonline.com/fr/biaxial-nema-17-bipolaire-0-9deg-44ncm-62-48oz-in-1-68a-2-8v-42x42x48mm-4-fils-17hm19-1684d
The important thing for the resolution is that they must be at 0.9° pitch.

polvinc, 2024-10-06:
What's certain is that I'll have to produce a more detailed doc, and provide the test results. The weather in France at the moment is totally unsuitable for astro photography.

Cepadues, 2024-10-06:
> As for the engines, I haven't bothered for the moment, I've taken two that I had in stock: https://www.omc-stepperonline.com/fr/biaxial-nema-17-bipolaire-0-9deg-44ncm-62-48oz-in-1-68a-2-8v-42x42x48mm-4-fils-17hm19-1684d The important thing for the resolution is that they must be at 0.9° pitch.

Are you satisfied with their current performances, or would you pick another one if you had to build more HEMYs ?

polvinc, 2024-10-06:
> Are you satisfied with their current performances, or would you pick another one if you had to build more HEMYs ?

Yes, they weren't chosen at random either. They have enormous torque and very satisfactory angular resolution. The only drawback is their weight. You could imagine making a slightly lighter mount by reducing the weight of the motors, especially if you wanted to carry less load. For my purposes, I won't be changing. The HEMY in this configuration weighs 5kg with its base, and can support >13kg according to my initial tests (15kg according to theory, 20kg with cw).

polvinc, 2024-10-06:
The next objective is to test HEMY's guidance performance. After adding shims to better hold the wave generator, I should potentially be able to go lower than the 1 arcsec I achieved this summer.
I also think that some gearboxes must be better than others. I'll have to test that one day.

Cepadues, 2024-10-06:
5kg including base is still light for an harmonic mount ! That's the same as an EQM-35 which can do 10kg MAXIMUM and guides far worse than your HEMY. 
I hope the weather in France becomes nicer ;)
Have a nice evening

martin0359, 2024-10-07:
Glad to hear you shared this, it basically solves most of my doubts. 
I am using a Chinese product s17-100 harmonic driver, which seems to have more load capacity than the Japan product and costs just under $50 USD, but I will test the actual results. 
Do these motors have to be planetary motors? I see in another project you used planetary motors.
For the AZ BASE, I plan to used a finished product that only requires replacing your quick release plate section.

martin0359, 2024-10-10:
Is there a way to convert rj45 to gx12 please? Unluckily my Onstep controller that I already own only have 2 gx12 ports. By the way what are the benefits of using rj45? If it's convenient, could you design a gx12 version for the T plate, my modeling skills are weak. This would help speed up my assembly process, much appreciated!

martin0359, 2024-10-10:
By the way, when will a stable version of Unicorn be released?

Cepadues, 2024-10-10:
You'd have to modify the T PLATE (5), GX12 just needs a round hole right ?

If so I could model it quickly from the plans in the repo, remind me on Sunday if you haven't heard of me.

polvinc, 2024-10-16:
> Do these motors have to be planetary motors? I see in another project you used planetary motors. 
No, here the reduction ratio is more than sufficient to dispense with an additional planetary gearbox (which would bring backlash).
> By the way, when will a stable version of Unicorn be released?I can also help with testing if I can
A few months from now. I'm working on other things at the same time, and we also have to make progress on the electronics for the HEMY brake.
> Is there a way to convert rj45 to gx12 please? Unluckily my Onstep controller that I already own only have 2 gx12 ports. By the way what are the benefits of using rj45? If it's convenient, could you [...]
The simplest solution is to make your own cable, otherwise you can 3D print or machine a plate to replace the Top Plate with the connectors you like.

martin0359, 2024-10-16:
> You'd have to modify the T PLATE (5), GX12 just needs a round hole right ?
> 
> If so I could model it quickly from the plans in the repo, remind me on Sunday if you haven't heard of me.

I think it will have 2 holes,thx.

Cepadues, 2024-10-16:
Here you go,

I assumed the diameter of the holes were 12 mm (from the name of the connector).
Be aware that the holes are untapped, and I'd recommend printing it before sending to a CNC shop to check dimensions are correct.

[Archive.zip](https://github.com/user-attachments/files/17400891/Archive.zip)

This issue is turning into a conversation, maybe it could be moved to a discussion instead ?

martin0359, 2024-10-18:
THX Cepadues
Thanks so much for avoiding me making a gx12 to rj45 cable!
This option is flexible

martin0359, 2024-10-18:
> Here you go,
>
> I assumed the diameter of the holes were 12 mm (from the name of the connector). Be aware that the holes are untapped, and I'd recommend printing it before sending to a CNC shop to [...]

I agree with that

closed 2024-12-03 by polvinc