polvinc/HEMY #4: Updated design?

chunhaow, 2024-11-19:
Dear Polvinc,

Thank you so much for this great design. Just wondering if you could release the updates/modifications you made to the mount and the base. I plan to CNC the parts and build one soon, and it would be great if I didn't build the outdated version. For the electronics, I can 3d print a top plate and use other solutions. Thanks a lot!

polvinc, 2024-11-19:
Hi, there,
Thanks for your interest in HEMY. I don't have time to commit all the updates yet. One or two more weeks of patience, is that OK with you? I'm still reviewing the electronics and mechanics before publishing.
But here's what it looks like :

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

chunhaow, 2024-11-19:
Thank you so much for the prompt reply! This looks great and promising! Looking forward to your commits.

polvinc, 2024-12-03:
I've just pushed the HEMY 1.1 upgrade, if you're interested in getting started. Mind you, even though I've done a detailed review, I haven't yet received my parts, nor the electronics. I should receive everything next week. So the work is not yet totally validated!

chunhaow, 2024-12-04:
Thanks a lot! I plan to CNC the parts soon, but have a few questions:

1. I saw the base parts have 2D drawing. I am wondering if you could also provide the 2D draws for the mount parts. Without these draws, they cannot figure out the diameters of the holes and where to tap threads, etc.

2. For the PCB, I can just build it from https://oshwlab.com/lordzurp/teenastro-redux right?

3. But what about the brake driver. Is it available somewhere?

Thanks again!

polvinc, 2024-12-04:
You're right, I haven't finished the “clean” drawings for the mount yet. For my machining, I did it case by case. Properly detailed drawings shouldn't take too long to be pushed onto the git, I'm on it.

As for the electronics, the TeenAstro Redux project we finished with LordZurp is not yet officially published, since we haven't yet been able to really test the electronics (especially the Brake Driver part). Nevertheless, I've put the Gerbers on the git (TeenAstro Redux, Brake Driver, and Smart Hand Controller). At your own risk if you start manufacturing before our tests! I made the PCBs with JLCPCB. Since I'm receiving my batch of pcb's by this WE, we should be releasing the official project shortly.

polvinc, 2024-12-04:
> 1. I saw the base parts have 2D drawing. I am wondering if you could also provide the 2D draws for the mount parts. Without these draws, they cannot figure out the diameters of the holes and where to tap threads, etc.

--> Done

closed 2024-12-04 by polvinc