polvinc/HEMY #5: 3d printed instead of CNC?

Trigger-broom-289, 2024-12-04:
Your project has to be the best open source harmonic drive mount I've seen so far.
Do you think the parts which are supposed to be aluminium CNC could be 3D printed? I do have a set of these harmonic drives.

polvinc, 2024-12-04:
I don't really have an opinion on that. In fact, I designed HEMY after DHEM because I thought there was too much bending in the 3D printed parts. So HEMY was designed from the outset to be optimized for CNC, and not for 3D printing. After that, I don't think it's unreasonable to try 3D printing. Just make sure you tweak the infill and other printing parameters to get the most rigid parts possible! Personally, I've 3D printed several versions of HEMY, but that was at the design stage, with draft parameters for printing.

I'm curious to hear if you achieve satisfactory results, let me know!

Trigger-broom-289, 2024-12-04:
I'm going to give it a try, might tweak a few inner parts if I see a weakness.

closed 2024-12-08 by polvinc