polvinc/HEMY #6: Questions for parts

chunhaow, 2024-12-05:
Thanks again for pushing the updated design. I have a couple of questions regarding the parts:

1. What's this marked part in the following picture? I didn't find a label for it. It looks like the wave generator of the harmonica drive, but it doesn't have the extrusion like in the second picture.
[image]

This is the picture of the little extrusion on the wave generator of HBS-17-100
[image]

I am trying to make sure I am getting the right harmonica drive.

2. How is the handle made? Is it CNCed or can you buy one from somewhere?

Thanks!

polvinc, 2024-12-05:
This is the wave generator. As my design works with HBS and CSF, the drawing is not perfectly representative. But in fact it can be fitted and adjusted! Regarding your image, it turns out that you can turn the wave generator the other way round so that it's flush with the pulley spacer. [image]

I had the handle in stock. You can buy it at farnell : [https://fr.farnell.com/mentor/3270-0883/poignee-acier-88mm-noire/dp/4216398](https://fr.farnell.com/mentor/3270-0883/poignee-acier-88mm-noire/dp/4216398). I forgot to put it in the BOM

chunhaow, 2024-12-05:
Thanks for the lightning reply! It makes perfect sense.

chunhaow, 2024-12-06:
One more question: what is the BTR 4 (M5) in the base BOM?

[image]

I couldn't find it in the assembly picture.

Already placed an order for the CNC parts. Now buying all other parts.

polvinc, 2024-12-06:
This is an allen key (BTR key) for adjusting the base (switching from GEM to Alt/Az, for example). The key can be placed in a special slot and held in place with a magnet.

[image]

I'm glad to hear that you're starting to make HEMY. Let me know if you have any problems with assembly.

chunhaow, 2024-12-08:
Thanks for identifying the Allen key! One more question:

You said the reduction ratio for DEC was 300:1. But you used 16T and 60T pulleys there. The reduction ratio is actually 3.75:1 instead of 3:1. Or do you actually mean 20T and 60T pulleys for DEC? In either case, is the length of the GT2 belt for this axis still 150mm?

polvinc, 2024-12-08:
Yes, I forgot to modify this part from the early designs... The pulleys are 16T and 60T for DEC. The reduction ratio is **3.75:1** instead of 3:1... **Many thanks for pointing out this (serious!) oversight on my part!** I'd have to change the info on the repo correctly.

I'm closing this issue to avoid adding too much to the diversity of related subjects. Don't hesitate to keep in touch, I'm always interested in feedback on your work!

closed 2024-12-08 by polvinc