polvinc/HEMY #7: Input voltage

chunhaow, 2024-12-08:
Do I have to use 24V input for the controller board because of the brake? Can I use 12V (because it's more versatile) if I can find a 12V brake or if use a 12-24 step-up? Do you recommend doing this? Thanks!

polvinc, 2024-12-10:
I haven't finished testing the electronics yet. The input voltage can be between 12 and 24V. The circuit will take care of motor and brake control, with the right voltages.

chunhaow, 2024-12-10:
Thanks! By further investigating the brake pcb, I saw there's a voltage regulator SDB628 which produces 24V output. So it's critical to buy a 24V brake.

polvinc, 2024-12-12:
I did read your message a bit quickly. Yes, you need to use a 24V brake. Also, I've just realised during the first hardware tests of the brake driver card that we made a big mistake upstream of the brake Enable control transistor: we need to add a 4.7k resistor in series... I'll be correcting the layout shortly

chunhaow, 2024-12-12:
Thanks for the update! Some updates on my side: I have ordered all the parts except the PCBs. I should be able to start building in a month.

It would be great to see the final version of the PCBs soon. Can you shed some light on building the PCBs? Do you solder all the components yourself or do you ask JLCPCB to do this? I found it very expensive to have JCL assemble a board, but on the other hand, I am not confident with my terrible soldering skills. Maybe with a stencil, solder paste, and hot air gun, it is manageable?

polvinc, 2024-12-15:
> It would be great to see the final version of the PCBs soon. Can you shed some light on building the PCBs? Do you solder all the components yourself or do you ask JLCPCB to do this? I found it very [...]

The cards (osntep and brake) are still being tested. But I hope to share them before the end of the week. Please note that the brake driver card currently available on github has a design error. I've fixed it, but haven't shared it yet. The same goes for the TeenAstro/OnStep board, on which we've moved the GPS connector that was touching the teensy.
As far as manufacturing is concerned, I have to admit that even though I have some soldering skills and a lot of equipment at home, I haven't started soldering myself. At least, not for most of the components. I soldered all the transverse components, but not the rest. We're still talking about a 5x5cm board with 2m of track and nearly 100 components to solder with nearly 400 small table stick pads (including M2 connectors, or 0402s, to solder and check with a magnifying glass).

closed 2024-12-24 by polvinc