polvinc/HEMY #8: Alternative control board

chunhaow, 2024-12-21:
Dear Polvinc,

While waiting for your final PCB designs, I couldn't resist thinking about the other modifications of the control board. TeenAstro is great on its own, but it's too expensive to build on JCLPCB (more than $110 USD), because of the number of components that they call "extended". I played around with EasyEDA and came up with a simplification: here I removed ESD protection, wrong polarity protection, and encoder from TeenAstro, changed TCM 2660 to TMC 2209, and used smaller sockets for stepper motors. Now the building cost can be reduced to around $60 USD. I am using the exact outline and connectors with TeenAstro to make sure it will fit in the mount.

Here are the schematic and the PCB picture. Not sure if this whole thing makes sense or I am making some obvious mistakes in the design (first time designing a circuit). Your feedback would be much appreciated!

[image]

[image]

chunhaow, 2024-12-21:
Just realized that your latest pcb is quite different from the files on TeenAstro Redux available here https://oshwlab.com/lordzurp/teenastro-redux. It would be great if you could also share your schematic so that I can make my modifications based on the updated design. Thanks a lot!

polvinc, 2024-12-23:
The updated version of our pcb will be arriving shortly, all we need to do is push it onto oshwlab correctly with the latest proofreading.

chunhaow, 2024-12-24:
> If $110 is too expensive, with all the protection that goes with it, more than 2,500 hours of development by three people whose job it is for some, layout optimisation, thinking about each component (basic or extended), all to support a mounting system that will bring a setup to several thousand euros, I don't really understand your logic. 
> All I can say to you is point you in the direction of the onstep forum for your pcb design. But in my opinion, it's far too dangerous and unprotected. Also, what's certain is that the GPS on your layout is going to have trouble picking up its signal (look at layout courses for RF). What's more, the capacitance reserves on the 100uF are strangely placed (they guarantee good power control of the motors). Maybe it'll work, probably it'll be fine, but imagine that it doesn't work the first time: then you're back to $60 (or much more) in engineering. If you want to get into development, I encourage you to do so, it's fascinating, but be careful not to get into the wrong battle: $110 is a small price to pay for development.

Sorry. When I commented "it was expensive to build", I didn't mean to undervalue you and your team's contribution and effort. I truly appreciate your effort to design and build every aspect of this excellent mount. Behind every github and oshwlab commit, there must be countless hours sitting in front of monitors for the mechanical/electronic/firmware designs/debugs and countless hours dealing with the PCB for testing! No value can match what you guys did and what you are sharing with us. All I was saying about the cost is for what I pay JLCPCB for the manufacturing. But I understand that I cannot expect much: this is a DIY project, and the cost cannot be matched with a mass production board.

More importantly, I had a lot of fun doing this. I have learned a lot these days studying circuit and PCB designs, functions of components, and how MCUs and drivers communicate in Onstep. It would be fantastic to see my modifications work eventually (even after a few failures, and I am ready for this). So it's not all about cost at this point.

Thanks a lot for pointing out the errors in my first attempt. I have learned so much from your every single reply. Looking forward to the updated design so that I can make some final corrections of my modifications.

chunhaow, 2024-12-26:
Thanks a lot for uploading the updated design on oshlab! I have made further modifications based on your latest design. I included the protection part and removed the encoder. Then I realized that I didn't need 5v at all because I could feed SHC using 3.3v. So I also removed the 40v->5v converter part! I spent a lot of time studying how MCU is communicating with drivers in onstep (especially from MaxPCB4), and then I rewired the driver by adding some jumpers to support more drivers with legacy/SPI/UART modes.

**Minor questions**: (1) is there any reason for using different inductors and capacitors in the buck converter circuit than what the datasheet recommends? For example, your L3 is 10uH, and your C34, C34 are 10uF; however, the datasheet of AP63203WU-7 recommends using 4.7uH and 22uF for them. (2) your L2 and L3 have a satuation current that is only 880mA. Would this be sufficient for other parts using 5V or 3.3V? Picture attached:

[image]

Here are my designs. I will do a final review and then send it to JLCPCB. Fingers crossed. Thanks again for the excellent and excellent designs!

[image]

[image]

polvinc, 2024-12-28:
Your design is nice. However, several recommendations: your Teensy MicroMod will touch the GPS component (I made this mistake on the PCB on HEMY's GitHub). You need to pay attention to the 3D model of your complete assembly. Our PCB on oshwlab has corrected this error. As far as the choke and capacitor are concerned, we've adopted a design that we use in our respective companies for step-downs. This design is normally sufficient for the purpose.
Be careful with your motor drivers too, you need to make sure they fit into HEMY, as they take up more space than our original design, and will need to be cooled with a heatsink.

lordzurp, 2024-12-31:
Hi
I did a batch with single 3v3 power for uC and SHC, and when I plugged the
SHC, it make a reset on uC ... That's why I go back to separate 5V for SHC
For the stepper driver, it's cheaper to integrate them on the pcb than buy
discrete driver and it's cleaner integration (IMHO)

lordzurp, 2024-12-31:
Extensive answer 

- AP6320x designs 
Forked from my company's standardized design, to be compatible with my production line's internal BOM.

- Board's logic consumption: around 300mA

- SHC comsumption: 150mA with 2"42 OLED display

chunhaow, 2025-01-01:
Hi polvic and lordzurp,

Thank you so much for the additional feedback! It's very helpful.

I do have an additional question regarding the GPS connection. I am not sure if you are able to find satellite signals in practice easily, but it looks like the impedance of the track connecting the RF port doesn't have the required 50ohm value. I used the JLC impedance calculator here https://jlcpcb.com/pcb-impedance-calculator and actually, it's impossible to match the 50ohm impedance in a 2-layer PCB (it requires a trace of width 2.7mm). But with a 4-layer PCB (with inner GND layers), it's quite easy: depending on the stack, 0.15-0.35mm width trace would be good.  Is your PCB made in 4 layers?

Also, I noticed a (possibly) error in the PCB design: the sizes of the holds of the M2 connector don't match the component in the BOM. The correct diameters should be 1.6mm and 1.1mm, respectively.
[image]

It's quite fun customizing your great designs! Thanks again! I noticed there's a spare serial port, so I added a header for HC05 Bluetooth (why not). I also removed the onboard GPS chip and used a header to connect to an external GPS module (e.g., https://www.aliexpress.com/item/1005006828932372.html or https://www.aliexpress.com/item/1005007883400656.html). Because of the uncertainty of matching the impedance, I'd play a safe game to get a stable GPS connection. I will have to enlarge the antenna window on the panel and customize the WIFI/Bluetooth and GPS holders.

This is the design I sent to JLCPCB. I will pay attention to the clearance above the stepper driver and the reset problem when plugging in SHC. I will update you once I receive the parts and electronics (in 2-3 weeks). Thank you again!

[image]

truongtu0191, 2025-01-04:
any plan to release a commercial ready board ?

polvinc, 2025-01-08:
> any plan to release a commercial ready board ?

For the moment, there are no plans to do so.

danb35, 2025-01-27:
I just quoted out five copies of the main board (based on what's in the repo right now) at JLC, and it came to a little over US$200 (so, about US$40 each).  But this really is begging for some kind of a group buy, because the "standard" PCB assembly (which is required due to some of the part choices) has some significant setup costs (not to mention shipping).  And I expect, though I haven't quoted this out yet, the same is true of the CNC parts.

chunhaow, 2025-01-29:
> I just quoted out five copies of the main board (based on what's in the repo right now) at JLC, and it came to a little over US$200 (so, about US$40 each). But this really is begging for some kind [...]

Some tips to reduce the cost: the parts that prevent economic processing are the power management IC DPS1135-FIA-13, and the center SMD nut. If you are able to solder these two by yourself, the cost will be significantly reduced. The SMD nut is very easy to deal with, while DPS1135-FIA-13 is difficult. You need to use a toothpick to apply a tiny amount of soldering paste to each soldering pad (0.5mm pitch), carefully place the IC, and then use a hot air gun to solder.

polvinc, 2025-02-02:
I agree with chunhaow. We designed this card to be ‘universal’. Not everyone has a €1000 soldering station and suitable equipment at home (I do!). For the time being, maybe one day we'll work out a system for distributing individual cards with a commercial partner, but that's not planned yet.
As far as the mechanics are concerned, it's true that the unit price goes down if you order several HEMYs, and that makes financial sense. That's what I did with people around me who were interested. But you still need to have people around you who are interested in embarking on this project!

The PCB is fully functional and has been tested. I'm very happy with it. I'll have to share the onstep config.h one of these days. The only thing that's left with this electronic board is a bit of operating noise on the motors that's a bit disturbing when tracking, perhaps linked to the TMC2660, which has the good taste of passing a lot of current, but doesn't have on-board silence algorithms like the TMC5160 or TMC2130.

chunhaow, 2025-02-15:
I am happy to report that my board works great. Wifi, Bluetooth, GPS are working. TMC2209 works silently and smoothly in uart mode. I added pull-ups to the enable pins of the stepper drivers, to prevent the brake from getting released during OnStepX is starting up. I like Bluetooth a lot. With it, I can control the mount with my phone without disconnecting from the internet.

My tripod arrives next week. So I can get my telescope on it then and do more tests.

[image]

polvinc, 2025-02-16:
I'm really pleased that you've managed to get a functional card that meets your expectations! Good idea, the pullups, I'll think about integrating them. As you can see, I'm going to go with your idea of integrated drivers, at least for a while. This will be to validate the rest of the pcb from top to bottom. 

I'm very happy you were able to do this, and I'm very interested in getting your feedback as you test. I don't have infinite time to develop designs and test them fully, so your feedback is invaluable.

chunhaow, 2025-02-19:
Hi @polvinc and @lordzurp , I am just wondering if you can receive signals from your GPS antenna. The track width out of your RF port doesn't match the required 50 ohm impedance, so in theory, it's not going to work. But on the other hand, it's just a few millimeters, and it might be fine. I am wondering how things go in practice vs in theory.

polvinc, 2025-02-24:
It's not easy to make a 50 Ohm track on a double layer. I played around with SaturnPCB to find the best configuration. Anyway, it's such a short journey that it doesn't really matter. All the clock cards I've seen being designed at work didn't take that much care over their track impedance when the path to the antenna was so short. 

I widened the track by around 1mm to get it closer to the desired impedance. And we're getting good GPS reception with a small external aerial.

What pull-up values did you use?

chunhaow, 2025-02-25:
It's good to know that the impact of impedance is not significant for such a short distance. One option to get the 50ohm impedance is to use a 4-layer PCB, and the manufacturing cost is still the same with JLCPCB.

I should have used 4.7k for the pull-ups as in MaxPCB4 and MaxESP4, but I don't have them in stock, so I used 5k instead.

closed 2025-04-08 by polvinc