polvinc/HEMY #9: Harmonic drive ratio

Trigger-broom-289, 2024-12-23:
Hello,

I currently have a couple of 17-30 instead of the 17-100, would that work or do I have to get the 17-100 ratio model?

thanks

polvinc, 2024-12-23:
I advise you to find out about reduction ratios, in relation to the angular resolution of the mount. In your case, a reduction ratio of 1:30 in the main with 1:5 in the pulley (for RA, 1:3.75 for DEC) is a bit tight in terms of real resolution. It will work, but you will have between 1 and 1.5 arcsec of resolution, which is a bit tight for astrophotography with long focal lengths.

[image]

Trigger-broom-289, 2024-12-23:
Thanks, I'll see if I order some new harmonic drives.

closed 2024-12-23 by Trigger-broom-289