polvinc/HEMY #11: Confusion regarding bom

Kerman-ap, 2024-12-29:
I see two versions of the BOM, the one in the readme and the one in the code files as separate ones for the mount and the base, am I right in saying it's better to follow the BOM in the code files? And would you call those complete?

polvinc, 2025-01-08:
There are indeed several BOMs. In fact, the ones in the PDFs of the frame and base plans were used for the design. The one in the table corresponds more to what I bought to make my HEMYs. It seems to me that the PDFs are more complete than the one in the readme. On the other hand, the one on the readme allows you to buy the parts directly.
I'm not sure if I'm being clear enough...

danb35, 2025-01-29:
If the BOM in the PDFs is definitive, I could submit a PR to update the docs (probably in its own file rather than in the README itself).  Would that be helpful?

danb35, 2025-02-01:
I'm trying to put together an updated BOM, but I'm running into a few questions:
* The README lists a few parts that I don't see in either of the PDFs--not for the mount or for the base.  These are the M6x54 clamping knob/screw, the two M6x50 clamping knob/screws, and the M8x14xThick adjusting shims.  Have these been removed, or am I missing them?
* Motors.  The README links to StepperOnline.com and calls for two identical motors (their model # 17HM19-1684D).  The PDF for the mount lists two separate motors, noting "double shaft" one one but not the other, and a weight difference of ~100g between them, but lists the same model number for both (which is the same model the README links to).  Is this right, or should the one designated D09 have a different model number noted?
* The handle.  I haven't been able to find an equivalent to the Mentor 3270.0883 on Aliexpress, but maybe I'm not searching for the right thing.  Are you aware of an alternative?
* The Vixen/Losmandy dovetail clamp.  Is there a particular part you had in mind, or can I just pick one more or less at random from Aliexpress and expect the screw hole pattern to line up?
* What about a counterweight shaft?  I recognize it won't be needed for most use, but is this designed to fit a particular off-the-shelf shaft?
* The SHC.  I see you've redesigned the PCB for it compared to the one in the TeenAstro project, but I assume it still needs a Wemos D1 Mini v4.0, along with the 2.4" OLED display?  Theirs calls for an I2C display, but the seven-pin connector you call for makes me wonder if your design needs SPI instead.  Though from the STEP file it appears you're only using a 4-pin header to connect to the display.
  * It also looks like your design is missing three buttons compared to theirs: Shift, f, and F.  Are these implemented differently?  Are you using different code for the SHC?  Or am I just missing them?
  * Is there an enclosure model available?  I presume the idea is that it would be 3D-printed.

This looks like a great project, and I'm looking forward to getting started (and hopefully helping with the docs some).

chunhaow, 2025-02-01:
> I'm trying to put together an updated BOM, but I'm running into a few questions:
>
> * The README lists a few parts that I don't see in either of the PDFs--not for the mount or for the base.  These are [...]

The clamping knob/screws in the PDF are good. The M8x14 shims are no longer needed in v1.1 (I think).

> * Motors.  The README links to StepperOnline.com and calls for two identical motors (their model # 17HM19-1684D).  The PDF for the mount lists two separate motors, noting "double shaft" one one but [...]

The DEC motor doesn't have the double shaft. Only the RA motor does (for the brake).

> * The handle.  I haven't been able to find an equivalent to the Mentor 3270.0883 on Aliexpress, but maybe I'm not searching for the right thing.  Are you aware of an alternative?

The handle is here: https://fr.farnell.com/mentor/3270-0883/poignee-acier-88mm-noire/dp/4216398 or here https://www.newark.com/mentor/3270-0883/handle-steel-88mm-ctrs-black/dp/15J5728 if you are in the US. I tried hard to find an alternative but failed...

> * The Vixen/Losmandy dovetail clamp.  Is there a particular part you had in mind, or can I just pick one more or less at random from Aliexpress and expect the screw hole pattern to line up?

Not sure what Polvinc got. But I ended up getting something else with matching holes.

>   * It also looks like your design is missing three buttons compared to theirs: Shift, f, and F.  Are these implemented differently?  Are you using different code for the SHC?  Or am I just missing them?

I haven't looked very deeply at SHC at this point. But according to what I have seen, you will loose some functions related to using f and F buttons with this SHC. If you want to use an SHC with full functionality, you probably need to use another 
SHC PCB while changing the ST4 connector to the 3.5mm audio jack connector. On the other hand, I plan to make another alternative to the HEMY board to use ST4 instead of 3.5mm jack.

Other possible errors in the BOM:
1. Two M5x14 hex bolts are missing in the base BOM, which are used to fix the Setting Alt to the Rotation Alt.
2. The RA belt should be 180mm rather than 200mm in the PDF BOM.
3. I found that using the 150mm belt on DEC is a bit tight and the motor doesn't fit without sanding the external shaft bearing retaining plate. I ordered an 154mm belt. Will see if it works better.

danb35, 2025-02-01:
> here https://www.newark.com/mentor/3270-0883/handle-steel-88mm-ctrs-black/dp/15J5728 if you are in the US

I'd hoped to not spend $10 in shipping for a $7 part, but I guess it can't be helped.  I guess the thing to do is to add more to the order, to spread out the cost, right?  Though it looks like their lead time is crazy.  I guess the handle isn't that critical a part, so I can add it after the fact if I need to.

> The DEC motor doesn't have the double shaft.

Are there any other particular characteristics I need to look for?  Or just NEMA 17, single shaft, 0.9°/step?

chunhaow, 2025-02-02:
Well, You will have to get the required rocker switch for the PCB from Newark/Farnell: https://www.newark.com/multicomp/4us1r102m7rn/switch-rocker-spdt-20v-black/dp/95M2277

So it makes sense to get the handle from them as well... I tried to find other sources for the switch, but couldn't find any.

Regarding the DEC motor, no other characteristics, just NEMA 17 single shaft, 0.9 deg/step should be fine. Also pay attention to the motor length. The BOM specified 47 or 48mm. I think slightly shorter motors would also work: the DEC motor doesn't need as much torque as the RA one.

polvinc, 2025-02-02:
Thank you chunhaow for your thoughtful replies, and for being more responsive than me!

Here are the ones I had in mind for the Vixen/Losmandy dovetail clamp. Now you need to look at the manufacturing drawing to make sure that the hole pattern matches the clamp you want. Another solution is to design another Harmonic/Vixen plug based on the existing one to fit the clamp you want:
[That one](https://fr.aliexpress.com/item/1005006483865820.html)
[This one](https://fr.aliexpress.com/item/1005005872254638.html)
Or the QR011 from Neewer.

For the DEC engine, you don't need a double shaft. I had this in stock, so I used it! Any Nema 17 with 0.9°/step will do. I've also designed the motor support to accommodate a Nema 14, but I haven't tested it.

polvinc, 2025-02-02:
> If the BOM in the PDFs is definitive, I could submit a PR to update the docs (probably in its own file rather than in the README itself). Would that be helpful?

Given that I haven't been able to be very reactive and free up time on this project in recent weeks, I'm actually looking for some help to make the readme more up to date and adjusted (from memory there are still a few errors in the BOM table in the readme). I'd also like, one day... to have some sort of assembly guide, like I did for DHEM.

danb35, 2025-02-02:
> I'm actually looking for some help to make the readme more up to date and adjusted

I'm not sure how much help I'll be on the rest of the README, but I'll try to have the BOM ready today.  Assembly guide would be nice, but I think can be figured out from the drawings (though I may yet eat those words, as I haven't tried it yet).  I'd think the bigger question would be for the firmware--there's lots of reference to TeenAstro, but it looks like their "flasher for dummies" only works with their boards and two stepper drivers, neither of which is the 2660.  So some notes on how to flash whatever needs to be flashed to wherever would be helpful.

danb35, 2025-02-02:
#13 submitted.

polvinc, 2025-02-02:
Thanks for this work. I reviewed and merged !

closed 2025-02-15 by polvinc