polvinc/HEMY #15: U1 doesn't fit main board

danb35, 2025-02-03:
I ordered a set of PCBs from JLCPCB.com using the Gerbers, BOM, and CPL files in this repo for v1.1.  This morning, I had an email from them saying that U1 (the m.2 connector) doesn't fit the PCB as designed--its locating pins don't match the holes in the board:
```
About the U1, the indicated pin is larger than the hole, it can't match, how about we cut the pin off? There is extra cost.
```

[image]

Here's the mounting footprint from the datasheet:

[image]

Link to the part:
https://www.lcsc.com/product-detail/Hard-Disk-Connector-SAS-SATA-M-2_FOXCONN-AS0BC21-S48BE-7H_C2761457.html

I'm not too concerned about the nominal extra cost, by itself.  And presumably they're still able to align the connector properly.  But it seems like a bug in the design.

chunhaow, 2025-02-03:
I think this is a bug. The holes on the PCB are too small to fit the m.2 adapter. See my post here: https://github.com/polvinc/HEMY/issues/8#issuecomment-2566800761

You need to edit the source file it using easyeda.

polvinc, 2025-02-03:
Outch, good find. We had this problem with the first batch of cards, and the drawing defect should have been corrected in the online version. Sorry for the sloppiness.
You have to use the PCBs from the online version: [https://oshwlab.com/lordzurp/teenastro-redux](https://oshwlab.com/lordzurp/teenastro-redux) and edit the footprint to put the correct diameters on the M.2 connector. I'm going to put a readme to that effect in the PCB folder, while [lordzurp](https://oshwlab.com/lordzurp/works) fixes the problem online.

polvinc, 2025-02-03:
@danb35 Which gerber did you take? The one on github, or the one on oshwlab?

danb35, 2025-02-03:
> Which gerber did you take? The one on github, or the one on oshwlab?

I used the v1.1 gerber from this repo.  Are those not usable?

polvinc, 2025-02-03:
> > Which gerber did you take? The one on github, or the one on oshwlab?
> 
> I used the v1.1 gerber from this repo. Are those not usable?

If it's still possible for you, I recommend using the pcb on [Oshwlab](https://oshwlab.com/lordzurp/teenastro-redux). The error on the M.2 connector is still there, but there has mainly been a correction on the positioning of the GPS module, and on the brake driver. If you use the PCB from the github (my bad, I haven't modified it), there will be a little soldering to do on the brake driver, and a washer to add to the main board.
I can guide you if necessary.

danb35, 2025-02-03:
I'll see if JLC can still cancel the order.

I'm not sure I'm comfortable enough with EasyEDA to make this change.  I think I've correctly edited the footprint for that connector, but I'm not sure how I'd go about applying that to the existing board layout.

polvinc, 2025-02-03:
With JLCPCB, you can also encourage them to apply the necessary corrections so that the M.2 footprint is correct.

Regarding the main PCB, on the gerber on github, be warned that you'll need to add an M3 washer between the pcb and the teensy micromod. One of the teensy's SOT23s is touching the GPS module, so it needs to be spaced out a bit (problem corrected on oshwlab). On the brake driver v1.0, the one on the github, there's simply an omission of a 4.7k resistor upstream of the activation transistor (problem corrected on oshwlab). If you build this version, you'll have to cut a trace and solder a 4.7k resistor to correct the problem. I can guide you if necessary.

danb35, 2025-02-05:
JLC canceled the main board but not the brake board, so I will need to mod those when they arrive.  I was able to modify the footprint on the main board, so that one should be fine.  And Sparkfun says they shipped the Teensy yesterday.

Now to get the CNC'd parts ordered.  Would you happen to know of a more economical source than JLC for shipment to .us?

danb35, 2025-02-10:
Raised https://github.com/zUrp-Astronomics/Unicorn/issues/1 on the upstream repo.

closed 2025-02-15 by polvinc