polvinc/HEMY #21: What is the part for P1 on the brake driver board?

danb35, 2025-02-15:
I'm trying to figure out what part I need to use for P1 on the brake driver board.  It mates with a 2x2 2.54mm male pin header on the main PCB, so I could use a female pin header, but it looks like that would leave the driver board sitting too high above the main PCB to be able to secure it in place using the same screw that holds in the Teensy.  There's a note on the PCB layout from EasyEDA saying `Order this part in your JLC inventory: WR-PHD SMT 62100421821`.  I'm not able to find any such part from LCSC or JLC.  It'd be nice, of course, if the PCBA could handle it, but it doesn't look like it'd be too hard to hand-solder if needed.  Any idea what this part should be, and where I can get it?

polvinc, 2025-02-15:
You have two options:
- Either you buy this part through your JLC account, choosing a supplier alernative to LCSC, and they automatically weld it during the SMT process. That's what I did by buying from Farnell via my JLC account.
- Or you can buy the part yourself and solder it yourself.

danb35, 2025-02-15:
My question was more what the part was, but the pointer to Farnell has me a lot closer.  But still not quite there; the closest I see is this: https://uk.farnell.com/wurth-elektronik/62130421821/conn-rcpt-4pos-2row-2mm/dp/3819101 -- one digit different in the part number.  But the datasheets at Würth (https://www.we-online.com/en/components/products/PHD_2_0_SMT_DUAL_SOCKET_HEADER_LOW_PROFILE_6210XX21821#62100421821) seem to indicate the only difference is that the 6213 part (which is what Farnell lists) comes on tape and has a plastic cap (which would presumably be needed for pick-and-place assembly), while the 6210 part comes in a tube and has no cap.

Is the 6213 part as equivalent as it appears, or am I missing something?

chunhaow, 2025-02-15:
You can get it from digikey or mouser, https://www.digikey.com/en/products/detail/samtec-inc/CLT-102-02-F-D-TR/6692113

danb35, 2025-02-15:
> digikey or mouser

Thanks.  I asked Würth for a few samples and will see what they say; otherwise Digikey will save a few bucks compared to Newark.

closed 2025-02-24 by danb35